ThomasBarth/ESP32-CAN-Driver #4: Maybe it does not work

wangsuyu, 2017-05-05:
Hi,ThomasBarth.
    Now I am workingt on ESP32-IDF，and I want to make can works.Thanks for espressif engineers recommand you to me.But it does not work when I make a test on your code.So,I want to make sure that have you ever test?If works,can you commit an example for me?Thank you very much.
Best Regards
Suyu Wang

santanapablo1975, 2017-05-05:
Hi

For a example take a look here

https://github.com/ESP32DE/ESP32-CAN-Driver/releases/tag/0.1

I made the test and it works very good!!!

Thanks Thomas! and Rudi!

Regards

Juan

wangsuyu, 2017-05-07:
Hi,Juan
Thanks for your response.Due to weekday,I do not have an esp32 to vertify.So,I spent several hours to read you code and leart a lot from yours.Thank you very much!
Thanks Thomas! ,Rudi! and @santanapablo1975 
Best Regards
Suyu Wang

wangsuyu, 2017-05-11:
Hi,juan
Sorry for a late update,I made some mistakes,so I just work your example well.
Thanks everyone!
Best Regards
Suyu Wang

closed 2017-05-17 by ThomasBarth