ThomasBarth/ESP32-CAN-Driver #15: undefined reference to CAN_cfg

physiii, 2018-01-29:
Having trouble compile - undefined reference to CAN_cfg:

> $ make
> CC build/can/CAN.o
> AR build/can/libcan.a
> CC build/main/hello_world_main.o
> AR build/main/libmain.a
> LD build/hello-world.elf
> /home/physiii/code/esp-examples/get-started/hello_world/build/can/libcan.a(CAN.o):(.literal.CAN_read_frame+0x0): undefined reference to `CAN_cfg'
> collect2: error: ld returned 1 exit status
> /home/physiii/esp/esp-idf/make/project.mk:388: recipe for target '/home/physiii/code/esp-examples/get-started/hello_world/build/hello-world.elf' failed

[Thomas' blog post](http://www.barth-dev.de/can-driver-esp32/) has this in [CAN_config.h](https://github.com/ThomasBarth/ESP32-CAN-Driver/blob/master/components/can/include/CAN_config.h) 
```
CAN_device_t CAN_cfg = {
   .speed=CAN_SPEED_500KBPS,
   .tx_pin_id = GPIO_NUM_5,
   .rx_pin_id = GPIO_NUM_4,
   .rx_queue=NULL,
};
```
Adding that to CAN_config.h in this repo gives:

> /home/physiii/code/esp-examples/get-started/hello_world/build/can/libcan.a(CAN.o):(.data.CAN_cfg+0x0): multiple definition of `CAN_cfg'
> /home/physiii/code/esp-examples/get-started/hello_world/build/main/libmain.a(hello_world_main.o):(.data.CAN_cfg+0x0): first defined here

Idea's on how to compile?

dpharris, 2018-01-29:
fgrep -nr CAN_cfg *

physiii, 2018-01-29:
> $ fgrep -nr CAN_cfg *
> Binary file build/can/CAN.o matches
> Binary file build/can/libcan.a matches

adding 
```
CAN_device_t CAN_cfg = {
   .speed=CAN_SPEED_500KBPS,
   .tx_pin_id = GPIO_NUM_5,
   .rx_pin_id = GPIO_NUM_4,
   .rx_queue=NULL,
};
```
to CAN_config.h gives

> $ fgrep -nr CAN_cfg *
> Binary file build/can/CAN.o matches
> Binary file build/can/libcan.a matches
> build/hello-world.map:14661: .data.CAN_cfg  0x0000000000000000       0x10 /home/physiii/code/esp-examples/get-started/hello_world/build/can/libcan.a(CAN.o)
> build/hello-world.map:19195: .data.CAN_cfg  0x000000003ffb1090       0x10 /home/physiii/code/esp-examples/get-started/hello_world/build/main/libmain.a(hello_world_main.o)
> build/hello-world.map:19196:                0x000000003ffb1090                CAN_cfg
> Binary file build/main/hello_world_main.o matches
> Binary file build/main/libmain.a matches

dpharris, 2018-01-29:
Do the fgrep on the source files ... does the sketch itself define  '
CAN_cfg'?

physiii, 2018-01-29:
Yes - I must be missing something basic.
My hello_world_main.c looks like:

```
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "CAN.h"

void app_main()
{
    printf("Hello world!\n");

    CAN_device_t CAN_cfg = {
     .speed=CAN_SPEED_500KBPS,
     .tx_pin_id = GPIO_NUM_5,
     .rx_pin_id = GPIO_NUM_4,
     .rx_queue=NULL,
    };

    CAN_init();
}
Yes - I have to be missing something very basic.
My hello_world_main.c looks like:

#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "CAN.h"

void app_main()
{
    printf("Hello world!\n");

    CAN_device_t CAN_cfg = {
     .speed=CAN_SPEED_500KBPS,
     .tx_pin_id = GPIO_NUM_5,
     .rx_pin_id = GPIO_NUM_4,
     .rx_queue=NULL,
    };

    CAN_init();
}
```
output:
> $ make
> CC build/main/hello_world_main.o
> /home/physiii/code/can_test/main/./hello_world_main.c: In function 'app_main':
> /home/physiii/code/can_test/main/./hello_world_main.c:10:18: warning: unused variable 'CAN_cfg' [-Wunused-variable]
>      CAN_device_t CAN_cfg = {
>                   ^
> AR build/main/libmain.a
> LD build/hello-world.elf
> /home/physiii/code/can_test/build/can/libcan.a(CAN.o):(.literal.CAN_read_frame+0x0): undefined reference to `CAN_cfg'
> collect2: error: ld returned 1 exit status

dpharris, 2018-01-29:
I think you need to make CAN_cfg global: 

#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "CAN.h"
CAN_cfg = {
    CAN_SPEED_500KBPS,
    GPIO_NUM_5,
    GPIO_NUM_4,
   NULL,
};
void app_main()
{
    printf("Hello world!\n");
    CAN_init();
}

dpharris, 2018-01-29:
Might have to be: 
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "CAN.h"

void app_main()
{
   printf("Hello world!\n");
   CAN_cfg = {
      CAN_SPEED_500KBPS,
      GPIO_NUM_5,
      GPIO_NUM_4,
      NULL
   };
CAN_init();
}

collin80, 2018-01-30:
That's the one. Of course you need the structure name too. I fixed that in
the below code sample. The code should compile as long as you define a
global variable called CAN_cfg