ThomasBarth/ESP32-CAN-Driver #17: Sending (writing) few frames in one go

martiinezz, 2018-03-06:
Hi, I have tried to send few frames (one after another) and it seems only first one makes it to the canbus after calling CAN_write_frame on each set of data (confirmed by another device listening only). Any hints how to do it properly?

sfowlr, 2018-03-19:
This driver currently doesn't block and wait until a transmit is completed before trying to send another, which may be the source of your issues. There's a fork that attempts to fix this issue at https://github.com/nayarsystems/ESP32-CAN-Driver/ - give that a try and see if it helps. I'd love to see these modifications pulled upstream into this repo, but nobody has put in a PR for it yet.

martiinezz, 2018-03-19:
I have found well working library here: https://github.com/collin80