ThomasBarth/ESP32-CAN-Driver #25: mainline in esp-idf ?

hackrid, 2019-07-26:
Hello,

thanks for your work on this!

what are your thoughts on including this into the main esp-idf project?
best regards!

collin80, 2019-07-26:
I don't think that will happen if only because the esp-idf has it's own CAN driver. So, in order to mainline Mr Barth's version you'd have to convince them to abandon their version and use this in it's place. This seems unlikely.

dpharris, 2019-07-26:
I had a look at 'theirs' and it looks pretty good.

https://github.com/espressif/esp-idf/blob/master/components/driver/include/driver/can.h
 and
   https://github.com/espressif/esp-idf/blob/master/components/driver/can.c

David

hackrid, 2019-07-26:
Hi fellows,

Thanks for your comments. 
For what ever reason I had the understanding that CAN is not supported in the esp-idf. 
My initial question is somehow solved then. 

Thanks!

closed 2019-07-26 by hackrid