ThomasBarth/ESP32-CAN-Driver #32: Payload length

nygma2004, 2024-02-13:
First of all thank you for your work. I am using this library to send data between ESP32 devices. All working fine so far, I just wonder why in CAN.h the u8 array in the CAN frame is defined as size 8? Am I mistaken that the a standard CAN frame has a size of up to 64 bytes? Or this was just choosen as an arbitary number when you created the library?

collin80, 2024-02-13:
A standard CAN frame can have 0-64 BITS of data in increments of 8. 64 bits is 8 bytes and thus the array is correct.

CAN-FD frames (which the ESP32 is not capable of producing) can have 64 bytes.

nygma2004, 2024-02-14:
Thank you. I probably looked at the CAN-FD documentation and did not realize the difference.

closed 2024-02-14 by nygma2004